redbear/STM32-Arduino #7: add ARDUINO_RedBear_Duo flag

hirotakaster, 2016-02-18:
Hi,
I want to add ARDUINO_RedBear_Duo define flag when compile on Arduno IDE.

STM32-Arduino/arduino/platform.txt
compiler.device.define=-DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_THREADING=1 -DPLATFORM_ID=88 -DPLATFORM_NAME=duo -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD058 -DUSBD_PID_CDC=0xC058  

to 

compiler.device.define=-DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_THREADING=1 -DPLATFORM_ID=88 -DPLATFORM_NAME=duo -DARDUINO_RedBear_Duo -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD058 -DUSBD_PID_CDC=0xC058  

When before pre-compile, ARDUINO_RedBear_Duo defined is added, but that flag is removed when compile.
I want to use this ARDUINO_RedBear_Duo flag for my MQTT library, please add this define to platform.txt if okay.

thank you.
Hirotaka

Cheong2K, 2016-02-22:
Hi, we will add PLATFORM=duo in the next board package and maybe release tomorrow.

hirotakaster, 2016-02-22:
Hi, "-DPLATFORM_NAME=duo" flag is already added when compile.
My point is 
- When precompile running, ARDUINO_RedBear_Duo flag defined as -DARDUINO_{build.board}  but PLATFORM_NAME=duo is not included.
- When compile running, ARDUINO_RedBear_Duo is not defined but PLATFORM_NAME=duo is included in compiler.device.define line.
so If there is no problem, please add -DARDUINO_{build.board} in compiler.device.define.
Thank you.

Cheong2K, 2016-02-23:
Hi, so we provide PLATFORM_NAME=duo for both will be ok?

hirotakaster, 2016-02-23:
Hi, okay, thank you!!

Cheong2K, 2016-02-23:
Hi, please use PLATFORM_ID=88 for the Duo, board package 0.2.4 is released.

hirotakaster, 2016-02-23:
Hi, thank you!! I will check new 0.2.4 version :)

Cheong2K, 2016-03-01:
Hi, does it work for you? Can I close this?

hirotakaster, 2016-03-01:
thank you!! it work well. I will close this issue.

closed 2016-03-01 by hirotakaster