redbear/STM32-Arduino #11: SimpleWebServerWiFi Fail

Goheeca, 2016-03-03:
When I was playing around with Duo and [this example](https://github.com/redbear/STM32-Arduino/tree/master/arduino/libraries/RedBear_Duo/examples/02.WiFi/SimpleWebServerWiFi), I came across a nasty bug.

I let the root page show on the Nexus 5 and that worked just fine, then I've used LG G3 instead of Nexus 5 and the Duo's begun blinking red. According to [Photon docs](https://docs.particle.io/guide/getting-started/modes/photon/#red-flash-sos) the LED indicated 'Hard fault'.

### How to reproduce

I've cought the communication with Wireshark and [here](https://gist.github.com/Goheeca/918538cf42ac7e3b18d9) are the particular http requests. Possibly change the host ip address in them accordingly. The Duo runs 0.2.2 fw.

The working request: `nc -i 1 10.1.2.205 80 < Nexus_5`
The non-working request: `nc -i 1 10.1.2.205 80 < LG_G3`

JacksonLv, 2016-03-07:
I use some Android Phones to test it .When I use the browser of system or google browser,it doesn't have problem.But when using other browser , the problem appear.

Goheeca, 2016-04-02:
I've looked in the code. The problem is the fixed buffer length (in which the lines from the http request have to fit in) or more generally the way of processing the http requests. The buffer is 150 bytes long. And the longest lines in requests are:

- Nexus 5: 148 bytes
- LG G3: 158 bytes

Clearly, that's not good for these hello world examples.