redbear/STM32-Arduino #15: MDNS in Particle Build does not feature `setService()`

ScruffR, 2016-04-23:
When using Particle Build to flash [this sample](https://github.com/redbear/STM32-Arduino/blob/master/arduino/libraries/RedBear_Duo/examples/01.Basics/Duo_WebServer/Duo_WebServer.ino) to the Duo, you'll get a "file not found" message for `#include "MDNS.h"`.
For Particle Build you'd need to import the `MDNS` library and the include statement would then be `#include "MDNS/MDNS.h"`

But then you'll get another error message "`'class MDNS' has no member called 'setService'`".
Changing this to `addService(...)` the sketch builds.

XuGuohui, 2016-05-26:
Hi @ScruffR, in the latest examples, the MDNS folder doesn't exist, which would fix the header problem. And the `addService(...)` has been replaced with `setService(...)`. Please check:

https://github.com/redbear/STM32-Arduino/blob/master/arduino/libraries/RedBear_Duo/src/MDNS.h#L49

Kind regards.

JacksonLv, 2016-05-26:
hi @ScruffR :  Our MDNS library is base on [here](https://github.com/mrhornsby/spark-core-mdns/tree/v1.1.1).But we changed a little bit code of  it ,so you can try v1.1.1.

ScruffR, 2016-05-26:
Thanks for the update, but I just tried that and now I can't even build the sample with the workarounds outlined in my OP.
*(I know this repo is mainly targeting Arduino IDE, but the `#if defined(ARDUINO) ... #else ...` suggested to me it should also work with Build)* 

How would I build that demo in Particle Build now?
This is what I get with [this commit](https://github.com/redbear/STM32-Arduino/commit/0e62565342a293a5f19e22da57b356ad88eed498) of the demo
```text
duo_webserver.cpp:47:36: error: 'BLE_APPEARANCE_UNKNOWN' was not declared in this scope
 #define BLE_PERIPHERAL_APPEARANCE  BLE_APPEARANCE_UNKNOWN
                                    ^

duo_webserver.cpp:62:12: note: in expansion of macro 'BLE_PERIPHERAL_APPEARANCE'
   LOW_BYTE(BLE_PERIPHERAL_APPEARANCE), HIGH_BYTE(BLE_PERIPHERAL_APPEARANCE) 
            ^
duo_webserver.cpp:62:37: error: 'LOW_BYTE' was not declared in this scope
   LOW_BYTE(BLE_PERIPHERAL_APPEARANCE), HIGH_BYTE(BLE_PERIPHERAL_APPEARANCE) 
                                     ^
duo_webserver.cpp:47:36: error: 'BLE_APPEARANCE_UNKNOWN' was not declared in this scope
 #define BLE_PERIPHERAL_APPEARANCE  BLE_APPEARANCE_UNKNOWN
                                    ^
duo_webserver.cpp:62:50: note: in expansion of macro 'BLE_PERIPHERAL_APPEARANCE'
   LOW_BYTE(BLE_PERIPHERAL_APPEARANCE), HIGH_BYTE(BLE_PERIPHERAL_APPEARANCE) 
                                                  ^
duo_webserver.cpp:62:75: error: 'HIGH_BYTE' was not declared in this scope
   LOW_BYTE(BLE_PERIPHERAL_APPEARANCE), HIGH_BYTE(BLE_PERIPHERAL_APPEARANCE) 
                                                                           ^

duo_webserver.cpp:70:29: error: 'LOW_BYTE' was not declared in this scope
   LOW_BYTE(MIN_CONN_INTERVAL), HIGH_BYTE(MIN_CONN_INTERVAL), 
                             ^

duo_webserver.cpp:70:59: error: 'HIGH_BYTE' was not declared in this scope
   LOW_BYTE(MIN_CONN_INTERVAL), HIGH_BYTE(MIN_CONN_INTERVAL), 
                                                           ^

duo_webserver.cpp:71:29: error: 'LOW_BYTE' was not declared in this scope
   LOW_BYTE(MAX_CONN_INTERVAL), HIGH_BYTE(MAX_CONN_INTERVAL), 
                             ^

duo_webserver.cpp:71:59: error: 'HIGH_BYTE' was not declared in this scope
   LOW_BYTE(MAX_CONN_INTERVAL), HIGH_BYTE(MAX_CONN_INTERVAL), 
                                                           ^

duo_webserver.cpp:72:25: error: 'LOW_BYTE' was not declared in this scope
   LOW_BYTE(SLAVE_LATENCY), HIGH_BYTE(SLAVE_LATENCY), 
                         ^

duo_webserver.cpp:72:51: error: 'HIGH_BYTE' was not declared in this scope
   LOW_BYTE(SLAVE_LATENCY), HIGH_BYTE(SLAVE_LATENCY), 
                                                   ^

duo_webserver.cpp:73:36: error: 'LOW_BYTE' was not declared in this scope
   LOW_BYTE(CONN_SUPERVISION_TIMEOUT), HIGH_BYTE(CONN_SUPERVISION_TIMEOUT)
                                    ^

duo_webserver.cpp:73:73: error: 'HIGH_BYTE' was not declared in this scope
   LOW_BYTE(CONN_SUPERVISION_TIMEOUT), HIGH_BYTE(CONN_SUPERVISION_TIMEOUT)
                                                                         ^

duo_webserver.cpp:105:20: error: 'BLE_GAP_ADV_TYPE_ADV_IND' was not declared in this scope
   .adv_type      = BLE_GAP_ADV_TYPE_ADV_IND,
                    ^

duo_webserver.cpp:106:20: error: 'BLE_GAP_ADDR_TYPE_PUBLIC' was not declared in this scope
   .dir_addr_type = BLE_GAP_ADDR_TYPE_PUBLIC,
                    ^

duo_webserver.cpp:108:20: error: 'BLE_GAP_ADV_CHANNEL_MAP_ALL' was not declared in this scope
   .channel_map   = BLE_GAP_ADV_CHANNEL_MAP_ALL,
                    ^

[... 51 lines cut ...]

duo_webserver.cpp:326:7: error: 'class BLEDevice' has no member named 'setAdvertisementParams'
   ble.setAdvertisementParams(&adv_params);
       ^

duo_webserver.cpp:329:7: error: 'class BLEDevice' has no member named 'setAdvertisementData'
   ble.setAdvertisementData(sizeof(adv_data), adv_data);
       ^

duo_webserver.cpp:330:7: error: 'class BLEDevice' has no member named 'setScanResponseData'
   ble.setScanResponseData(sizeof(scan_response), scan_response);
       ^

duo_webserver.cpp: In function 'void loop()':
duo_webserver.cpp:380:38: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
         if (endsWith(buffer, "GET /H")) {
                                      ^
duo_webserver.cpp:383:38: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
         if (endsWith(buffer, "GET /L")) {
                                      ^
make[1]: *** [../build/target/user/platform-88duo_webserver.o] Error 1
make: *** [user] Error 2
```

Maybe it would be good to have a contributed "library" on Particle Build that allows users there to just pick a sample which builds out the box (or with explicit build messages to actively import required third party libraries if not bundled).

XuGuohui, 2016-05-26:
You can not compile the example on Particle Build now, as the build farm has been updated before v0.2.4 is available. We manage to release firmware v0.2.4 this week, but it still takes a bit short time to make it available on Particle Build.

>Maybe it would be good to have a contributed "library" on Particle Build that allows users there to just pick a sample which builds out the box (or with explicit build messages to actively import required third party libraries if not bundled).

That's what we exactly want to do, while we have limit time and resources to contribute the libraries. It'll come to developers sooner or later and any contributes from developers would really appreciated.

Thanks!

XuGuohui, 2016-05-27:
Hi @ScruffR ,

You can now build applications for Duo firmware v0.2.4 on Particle Build now! 

Cheers!

ScruffR, 2016-05-27:
I'll give it a try 
Thanks!